bespoke-silicon-group/basejump_stl #579: Syntax errors in generated hard-macro-swapping verilog files when using Python3

derekcom17, 2022-07-14:
How to observe the:
1. Navigate to `basejump_stl/hard/gf_14/bsg_mem`
2. Run the command `python3 ../../common/bsg_mem/bsg_mem_generator.py ../../common/bsg_mem/memgen.json 1rw 1 > mems.v`
3. Open the generated file `mems.v`

**The issue** can be seen on line 35:
```
`bsg_mem_1rw_sync_mask_write_bit_1rf_macro(512.0,64.0,m2) else
```
The ram dimensions have a decimal point in them. The verilog parser (or at least Genus which I was testing with) sees this as a syntax error.

This issue is not present if you run the file generator with Python2. After running `python2 ../../common/bsg_mem/bsg_mem_generator.py ../../common/bsg_mem/memgen.json 1rw 1 > mems.v`, the **expected output** is seen on line 35:
```
`bsg_mem_1rw_sync_mask_write_bit_1rf_macro(512,64,m2) else
```

derekcom17, 2022-07-15:
I think this line causes the problem: 
https://github.com/bespoke-silicon-group/basejump_stl/blob/2c75c26996a96b87e1738e6bc4f3144159cfca7b/hard/common/bsg_mem/bsg_mem_generator.py#L752
This "integer/integer" has different behavior in python 2 vs 3. Only in python2 does it appear to return an integer if it divides evenly. 

Also, the problem case was with Python 3.6.8 and the correct case was with Python 2.7.5 in case that makes a difference.

dpetrisko, 2022-10-18:
Fixed with #578

closed 2022-10-18 by dpetrisko