bespoke-silicon-group/basejump_stl #586: bsg_cache optimization opportunity

dpetrisko, 2022-10-01:
When bsg_cache has a miss. it fills the new data and then goes to a "recover" state, where it re-reads tag and data mems in TL. I believe this step can be omitted if v_tl_r == 0, since the memory data in that stage will be unused anyway. This has three major benefits:
- Saves power by not rereading the RAMs
- Saves a cycle of latency on miss
- Saves a cycle of latency on hit-after miss

dpetrisko, 2022-10-01:
I believe that the changes will be restricted to bsg_cache_miss.v

tommydcjung, 2022-10-02:
I guess the question is how often is TL empty when the cache is in miss. Probably not a common case.

dpetrisko, 2022-10-02:
I agree when using bsg_cache as an L2 with cacheline-oriented operations, but this is very common for an L1 with a random access pattern. I discovered this when analyzing the performance of a different cache and using bsg_cache as a reference