bespoke-silicon-group/basejump_stl #594: :bug: sync_len_lp is undeclared and undefined in config_snooper.v

flaviens, 2022-10-12:
Hi there!

In `config_snooper.v`, the signal `sync_len_lp` is neither declared nor defined, however it is used in line 26.

https://github.com/bespoke-silicon-group/basejump_stl/blob/0174e65eeb3f372085dbc013721ac08fd709546b/bsg_tag/legacy/config_net/src/config_snooper.v#L26

taylor-bsg, 2022-10-12:
thanks! I added a pointer for where this is defined.

https://github.com/bespoke-silicon-group/basejump_stl/commit/31551222a26c6545765b4efb6d1d7ce2dd7d5a9f

flaviens, 2022-10-12:
Aha! Interesting, my mistake. What led to the issue is that there are multiple `config_defs.v`, protected by ifdef/define snippets, and I included several of them, so in this context `sync_len_lp` wasn't defined, and I didn't look at the right place.
Would it make sense to implement checks to ensure that several `config_defs.v` are not included in the same scope?