bespoke-silicon-group/basejump_stl #605: bsg_cache.v ready_o may be yumi_o

BrendenPage, 2022-10-31:
https://github.com/bespoke-silicon-group/basejump_stl/blob/master/bsg_cache/bsg_cache.v#L980-L995

dpetrisko, 2022-12-21:
Hi @taylor-bsg @tommydcjung can we get some guidance on this? This is a handshake problem because if we connect bsg_cache to a demanding ready->valid module there will be a combinational loop. I see a few solutions:
- disable the tl_we if tag_op optimization and only ready if TL is available
- Change the interface to yumi_o and update _to_cache modules
- Define a new type of handshake _then_ready which is different than yumi by virtue of not acting as an ack but restricting the handshakes it attaches to be helpful 

My preference is probably for 1 as I don’t think tagst is ever used in a mixed workload where performance is important. Thoughts?

dpetrisko, 2023-01-21:
Resolved

closed 2023-01-21 by dpetrisko