bespoke-silicon-group/basejump_stl #609: Add code for FPGA side of bsg_link

taylor-bsg, 2022-11-15:
https://github.com/bespoke-silicon-group/basejump_stl/blob/321c89653f5ad7fb1236f18c3ade085e5dbbb9d3/hard/ultrascale_plus/bsg_link/README.md?plain=1#L1

Hi Paul, I noticed that we don't have the FPGA code for bsg_link -- do you think you can add it? I assume that as we build the HB daughterboard, this will be a component we use. This will also be very useful for external users of the bsg_link module.

taylor-bsg, 2022-11-15:
@gaozihou also would it make sense to have a FPGA bsg_tag component with STA constraints?

dpetrisko, 2022-11-22:
@gaozihou additional info that would be helpful: max speed of clock/data pins achieved on FPGA, and if there are any board-level tricks required for that

taylor-bsg, 2026-09-11:
| 33 | Issue [#609](https://github.com/bespoke-silicon-group/basejump_stl/issues/609) — Add code for FPGA side of bsg_link | Close original request; retain follow-up documentation | The UltraScale+ link code, integration README, and example placement/timing constraints are present. The later requests for measured FPGA speed, board guidance, and FPGA tag collateral remain documentation follow-ups rather than missing link RTL. [current source](https://github.com/bespoke-silicon-group/basejump_stl/blob/11d19a888daa24e8239fc81c9c397b4d7bd58568/hard/ultrascale_plus/bsg_link/README.md). |