bespoke-silicon-group/basejump_stl #637: Broken cache testbenches

BrendenPage, 2023-01-11:
It appears that somewhere along the way, several of the bsg_cache testbenches fell out of date with the cache implementation. These tests include `axe_test`, `dmc`, and `lock_test` where each are missing port connections that have been added since they had last been updated.

I do not understand the cache implementation enough to lend any significant help to the effort of fixing these, but if they are no longer important tests, we should remove them, otherwise we should try to keep them up to date.

taylor-bsg, 2023-01-12:
@tommydcjung @farzamgl can you take a look -- which are deprecated and which should be maintained

farzamgl, 2023-01-12:
The `dmc` testbench has been outdated for a while(before the word tracking changes), but I think I can get it running with a bit of effort. @tommydcjung any ideas about the other 2? Also, I think `axe_test` is currently not maintained.

tommydcjung, 2023-01-15:
I would say "regression", "regression_v2", "regression_64" are the must. Other ones are less critical, unless you are testing the other peripheral components with bsg_cache.

BrendenPage, 2023-01-23:
@tommydcjung Would you say that we should try to keep the testbenches other than the three regression testbenches current or would it not be worth it given the current state of the repo? I'd ideally like to propose that any testbenches we don't plan on maintaining be removed from the testing folder and possibly be either moved to some "testing legacy" folder or removed entirely.

tommydcjung, 2023-01-24:
I think the overall goal is whether we have good test coverage for every module provided in basejump stl. axe_test can be  removed, because other testbenches should be covering its test space pretty well. lock_test exercises alock features which are not well covered in regression_v2, but we might be able to merge them into regression_v2.  Then, there are things that attaches to the cache like

- bsg_cache_to_dram_ctrl
- bsg_cache_to_axi
- bsg_cache_to_test_dram
- bsg_wormhole_to_cache_dma_fanout
- bsg_cache_dma_to_wormhole

I think they should be maintained, because it provides test coverage for those modules. Some of the directories should be renamed to match the name of the module it's testing (e.g. axi_test -> bsg_cache_to_axi). Dan and Farzam should be able to help with updating those testbenches.