bespoke-silicon-group/basejump_stl #671: Getting Started with BaseJump STL

ShvetankPrakash, 2024-03-01:
I was wondering if there is any tutorial videos or "Hello World" examples for getting started with the library? Thanks!

dpetrisko, 2024-03-01:
Hi,

The paper and this video are good overviews of the repo. 

https://m.youtube.com/watch?v=7MsS3d81Wdw&list=PLUg3wIOWD8yoZx3W8JfSMn4tiHXELmgMT&index=16&t=377s&pp=iAQB

https://github.com/bespoke-silicon-group/basejump_stl/blob/master/docs/BaseJump_STL_DAC_2018_Camera_Ready.pdf

ShvetankPrakash, 2024-03-20:
Thanks! I have checked these out but was wondering if there are any open-source, concrete examples/designs that you could point me to for how people have pieced together modules from the repo.

dpetrisko, 2024-03-20:
Hammerblade manycore:
https://github.com/bespoke-silicon-group/bsg_manycore

BlackParrot multicore:
https://github.com/black-parrot/black-parrot

closed 2024-03-21 by ShvetankPrakash