bespoke-silicon-group/basejump_stl #672: bsg_dataflow/bsg_fifo_bypass.sv: Port/variable name mismatch

infinitymdm, 2024-03-01:
In bsg_dataflow/bsg_fifo_bypass.sv there is a variable `fifo_ready_param_i` that does not match up with any port name.

https://github.com/bespoke-silicon-group/basejump_stl/blob/14f29e4d8a498ad62ec05fe232407f9d887e36a9/bsg_dataflow/bsg_fifo_bypass.sv#L4-L36

By comparison with the [testbench](https://github.com/bespoke-silicon-group/basejump_stl/blob/14f29e4d8a498ad62ec05fe232407f9d887e36a9/testing/bsg_dataflow/bsg_fifo_bypass/testbench.sv#L46), I assume this is supposed to be `fifo_ready_i`?

dpetrisko, 2024-03-01:
Yep, the port name of the module should be changed. Feel free to PR!

infinitymdm, 2024-03-01:
Cool, just double checking. Expect a PR shortly.

PR #673 (2024-03-02): bsg_dataflow/bsg_fifo_bypass.sv: Fix incorrect port name
Fixes #672
closed 2024-03-02 by dpetrisko