bespoke-silicon-group/basejump_stl #679: bsg_dmc_s timing parameters

infinitymdm, 2024-04-24:
Hi folks,

My research group is trying to use bsg_dmc as a memory controller for a custom RISC-V SoC. I'm looking for clarification on what values to use for the timing parameters, as done here in the testbench:

https://github.com/bespoke-silicon-group/basejump_stl/blob/90bd83a69f94d123705befebd8b2a234166b0407/testing/bsg_dmc/testbench.sv#L115-L133

Of course, we're going to use config registers to do this in our SoC implementation. This question isn't about _how_ to assign these values, it's about _what values to use_. 

We're planning to use [this Micron part](https://www.mouser.com/ProductDetail/Micron/MT46H128M16LFDD-48-ITC-TR?qs=taEdVNyAfdH9Uc4RiwxL6g%3D%3D). Their datasheet specifies most of these values in microseconds or nanoseconds. **How do I translate these timing values into raw integers to configure the memory controller?**

For example, they specify a minimum tREFI of 7.8 us. What value should `dmc_p.trefi` be assigned?

dpetrisko, 2024-04-24:
I believe they are in nanoseconds for the most part. So for example dmc_p.trefi = 7800

You can see a Micron provided example here:
https://github.com/bespoke-silicon-group/basejump_stl/blob/90bd83a69f94d123705befebd8b2a234166b0407/testing/bsg_dmc/lpddr_verilog_model/128Mb_mobile_ddr_parameters.svh

BTW, we have a full LPDDR subsystem here ( including configuration and testing interfaces) here: https://github.com/bespoke-silicon-group/bsg_pearls/blob/main/bsg_dmc_pearl/bsg_dmc_pearl.v

dpetrisko, 2024-04-24:
Great, let us know how it goes and make sure to contribute back your hardened modules if you can! We have a 28nm implementation of the delay line here:
https://github.com/bespoke-silicon-group/basejump_stl/tree/master/hard/tsmc_28/bsg_dmc