bespoke-silicon-group/basejump_stl #680: bsg_dmc: are non-burst write operations supported?

infinitymdm, 2024-04-26:
Hi folks,

I'm trying to integrate bsg_dmc into a RISC-V SoC by translating its AHB bus into something compatible with the Xilinx UI. Because the transaction format for AHB includes both single-transfer and incrementing burst writes (and I mostly expect single writes), I'd like to know whether bsg_dmc supports non-burst write operations at all.

I notice that the testbench provided performs only 8-beat bursts, and does not test single transfers. Is this meant to imply that bsg_dmc does not intend to support single transfers? (i.e. do I need to add logic in to translate single transfers into burst transfers?)

infinitymdm, 2024-04-30:
The behavior I'm observing during testing is that after some number of non-burst write transfers, bsg_dmc stops issuing commands to memory and more or less hangs forever. Here's a snippet from simulation: 
[image]

In our test, we issue 10 single write transfers to the memory controller. You can see 8 of them write correctly to memory, but after the 9th is issued over the UI, `app_rdy` deasserts (as bsg_dmc's command fifo is full) and the controller hangs.

Is this behavior perhaps related to the issue mentioned in #387 (in which case I suppose we need to use the fixes in #584)?

infinitymdm, 2024-05-14:
The conclusion I draw from my testing is that bsg_dmc supports exclusively burst transfers. The only way I can find to get it to support non-burst transfers is to set `burst_data_width_p` to the same value as `ui_data_width_p`.

closed 2024-05-14 by infinitymdm

dpetrisko, 2024-05-15:
>I notice that the testbench provided performs only 8-beat bursts, and does not test single transfers. Is this meant to imply that bsg_dmc does not intend to support single transfers?

Yes, the controller only supports burst transfers. Generally in a processor you'll set your burst len to the LLC line size and the data size to the bus width

infinitymdm, 2024-05-15:
This limitation ought to be documented somewhere rather than just implicitly in the testbench, as Xilinx's UI spec permits non-back-to-back transfers.