bespoke-silicon-group/basejump_stl #694: commone interface stl

neilwang0913, 2024-09-07:
can you add some common interface stl such like i2c, spi etc?

dpetrisko, 2024-09-07:
Hi, these are lower level building blocks. We have higher-level, silicon-validated components at https://github.com/bespoke-silicon-group/bsg_pearls

neilwang0913, 2024-09-07:
Thank you for pointing out that. Is there any document about that? The name is not a standard one, which makes it difficult to understand it—many thanks.

dpetrisko, 2024-09-07:
The name is a pun :) They are intended to be hardened nodes within a softened mid-level design. For instance, an SoC I’m taping out currently has 10+ bsg_clk_gen_pearl, 50+ bsg_sdr_link_pearl, and some bsg_ddr_link_pearl which are stamped out as hard macros

The intention of the repo is to provide the RTL, constraints and bringup scripts all-in-one, but it is a work in progress to open-source some of these from internal tapeouts