bespoke-silicon-group/basejump_stl #714: Erroneous counter reset?

dpetrisko, 2025-01-06:
May be a verilator bug, have not jumped in. What's the intention of this code? The comment suggests this is error-prone, perhaps this is a real error...

```
%Warning-WIDTHTRUNC: /home/petrisko/scratch/pearls_explore/bsg_pearls/../basejump_stl/bsg_async/bsg_async_credit_counter.sv:92:26: Operator ASSIGNDLY expects 3 bits on the Assign RHS, but Assign RHS's REPLICATE generates 32 bits.
                                                                                                                                 : ... note: In instance 'testbench.dut.chip.sdr.uplink.sso.neg_credit_ctr'
                 // fixme? not sure this constant will always do as expected
   92 |        r_counter_r       <= { -max_tokens_p * start_full_p, { lg_credit_to_token_decimation_p  {1'b0} } };
      |                          ^~
                     ... For warning description see https://verilator.org/warn/WIDTHTRUNC?v=5.030
                     ... Use "/* verilator lint_off WIDTHTRUNC */" and lint_on around source to disable this message.
```

gaozihou, 2025-01-06:
@dpetrisko The intention is to reset the r_counter to `minus num_credits`, and reset w_counter to `0`. During normal operation r_counter (local counter) will count up, if r_counter == w_counter the credit is no longer available unless w_counter count up (credit returned).

I guess this warning is just saying there is no explicit width conversion from the 32-bit param to the 3-bit counter logic.. Maybe it is better to add an explicit width conversion

dpetrisko, 2025-01-06:
Hm, I see thanks!  I think this is saying that r_counter_r will reset to 0 since the replication part of the concatenation is 32b. But I'll check it out in a simulation and see what happens. It may be a verilator bug, since it seems like the replication should be lg_credit_to_token_decimation_p wide rather than 32. If it's a verilator bug, I'll raise it there!

gaozihou, 2025-01-06:
@dpetrisko `lg_credit_to_token_decimation_p` is `0` in your case, so it will not be rounded to 0, it should be rounded to `-4` (`0b100`)